joric/nrfmicro #6: Problem to unlock the nrf chip with Blackmagic

BenRoe, 2019-09-16:
I try to unlock the nrf chip on Linux Debian.
Port is correct (ttyACM0 and ttyACM1 are available).
Wirering should be correct (see picture).
Any idea what the problem could be? 

I can't figure out what the problem is. 

```bash
arm-none-eabi-gdb --batch -ex "target extended-remote /dev/ttyACM0" -ex "mon swdp_scan" -ex "att 2" -ex "mon erase_mass"
Target voltage: unknown
Available Targets:
No. Att Driver
 1      Nordic nRF52 Access Port
Attaching to Remote target failed
Target does not support this command.
```
[image]

I also get only one available target in gdb.
```bash
pi@raspberrypi:~/Downloads $ arm-none-eabi-gdb
GNU gdb (7.10-1+9) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=arm-linux-gnueabihf --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) target extended-remote /dev/ttyACM0
Remote debugging using /dev/ttyACM0
(gdb) set non-stop on
(gdb) mon swdp_scan
Target voltage: unknown
Available Targets:
No. Att Driver
 1      Nordic nRF52 Access Port
(gdb)
```

Lenbok, 2019-09-17:
@BenRoe Cool to see someone else making these.

Before the module is unlocked there is only one target available (the `Nordic nRF52 Access Port` that you see) and that is the one you need to attach to and do the `mon erase` on.

After that, when you connect or do the next `swdp_scan`, you should see two targets available. On my box (old PC hardware running Ubuntu 18.04), I found it a bit troublesome, in that the unlock would kill my USB port.

Here are a couple of pastes from the logs I've been keeping as I go... 

Side note: I set up some udev rules so the black magic probe always gets the same name:

```bash
cat <<EOF | sudo tee -a /etc/udev/rules.d/99-blackmagic.rules
# Black Magic Probe
# there are two connections, one for GDB and one for uart debugging
  SUBSYSTEM=="tty", ATTRS{interface}=="Black Magic GDB Server", SYMLINK+="ttyBmpGdb"
  SUBSYSTEM=="tty", ATTRS{interface}=="Black Magic UART Port", SYMLINK+="ttyBmpTarg"
EOF
sudo service udev restart
```
That lets me use `/dev/ttyBmpGdb` with gdb.

On to the unlocking:
```bash
$ arm-none-eabi-gdb --ex "target extended-remote /dev/ttyBmpGdb" --ex "mon swdp_scan"
GNU gdb (GNU Tools for Arm Embedded Processors 7-2018-q3-update) 8.1.0.20180315-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-pc-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
Remote debugging using /dev/ttyBmpGdb
Target voltage: unknown
Available Targets:
No. Att Driver
 1      Nordic nRF52 Access Port
(gdb) att 1
Attaching to Remote target
warning: while parsing target description: no element found
warning: Could not load XML target description; ignoring
PC register is not available
(gdb) mon erase_mass
(gdb) detach
Detaching from program: , Remote target
(gdb) quit
```
at which point I had to reboot, then:

```bash
$ arm-none-eabi-gdb --ex "target extended-remote /dev/ttyBmpGdb" --ex "mon swdp_scan"
[...]
Target voltage: unknown
Available Targets:
No. Att Driver
 1      Nordic nRF52 M3/M4
 2      Nordic nRF52 Access Port
(gdb) att 1
Attaching to Remote target
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0xfffffffe in ?? ()
(gdb) mon erase_mass
erase..
(gdb) load pca10056_bootloader-0.2.11_s140_6.1.1.hex
Loading section .sec1, size 0xb00 lma 0x0
Loading section .sec2, size 0xf000 lma 0x1000
Loading section .sec3, size 0x10000 lma 0x10000
Loading section .sec4, size 0x5de8 lma 0x20000
Loading section .sec5, size 0x70c8 lma 0xf4000
Loading section .sec6, size 0x8 lma 0x10001014
Start address 0x0, load size 182712
Transfer rate: 31 KB/sec, 966 bytes/write.
(gdb) detach
Detaching from program: , Remote target
(gdb) quit
$
```

Then power off BMP; power on BMP; power on USB to nrfmicro; mount the drive that appears. (Note that this was unlocking and flashing the raw module before connecting to the nrfmicro PCB, so I needed the BMP powered on to supply 3.3v to the module -- since yours is assembled you should just be able to connect it directly to the USB).

Now you can copy the precompiled test firmware to check each GPIO can flash a test LED as described on the wiki:

```bash
$ cp blink.uf2 /media/len/NRF52BOOT/
```
After the file copies, the volume will disappear, and the blink code is running, you can then proceed to verifying the pins on the nrfmicro. In all of the ones I soldered I had to wick some more solder to some of the inner pads to get the LED flashing, it's a neat design trick that @joric used there!

I've got my nrfmicros working with the test master/slave keyboard firmware on a corne, and am currently (when I can grab bits of time) updating to get the keymap I normally use working.

joric, 2019-09-17:
Great to see someone else experiences problems with my project! Also take a look at that https://github.com/AndruPol/nrf52832-recover/issues/1 (link from References in the [Software](https://github.com/joric/nrfmicro/wiki/Software) article).

closed 2019-10-20 by joric