joric/nrfmicro #8: Make 104 keyboard firmware

624245305, 2019-10-25:
Make 104 keyboard firmware, add PIN pin under custom_board.h and pin_assign.h file, make firmware, add some pins and no response.

NRF52 does not have a CAPS/NUM LED pair?

joric, 2019-10-25:
I'm not supposed to decode this.

joric, 2019-10-25:
Assuming that "104" means you're making a 104-key keyboard using my controller, you won't have enough broken out pins. You can solder your own though.
Assuming "CAPS/NUM LED pair" is about indicator leds (there are usually three of those), I'm really not sure, it may be lost somewhere in the nrf52 fork it could need reading through the code. I haven't really used those.

closed 2019-10-25 by joric

Lenbok, 2019-10-25:
I think you are right about the nrf52 fork losing caps/num indicator support, as it didn't work for me when I tried displaying those led status on the crkbd OLED a few days ago.