joric/nrfmicro #11: Battery life with coin battery?

Na-Cly, 2019-11-13:
Hey just wondering what kind of battery life you see with the coin cell battery. Do you know what it should be for a fully wireless split keyboard
Thanks.

joric, 2019-11-13:
I'd rather use 130 mAh 501230 lipo batteries they fit under promicro. Battery life could vary from 1-2 weeks to a few months but it doesn't really matter because batteries are rechargeable.

closed 2019-11-13 by joric

Lenbok, 2019-11-13:
Nice to see that battery recommendation. I am using these https://www.aliexpress.com/item/32981806740.html (they are quite long). 

I don't currently have a battery estimate (I'm still trying to find out why I can't pair with my linux box, when it pairs to MacOS and Android just fine, could be my BT dongle), but a couple of things I've noted so far:

* With the current firmware the blue nrfmicro LED is illuminated essentially all the time, so we probably need to switch that off in software for battery conservation.
* I'm not sure how to monitor battery voltage on the slave side (for the master side there is a binding to echo it out as a string of keypresses, and I've also got it displaying on the oled, but I only have one oled connected)