joric/nrfmicro #14: Reciever option

haywirephoenix, 2019-12-16:
Would it be straight forward to switch to proprietary 2.4ghz + receiver approach? Pros would be increased battery life, using keyboard in bios and wake from sleep.

joric, 2019-12-16:
This is not planned, I don't like receivers.

closed 2019-12-16 by joric

haywirephoenix, 2019-12-16:
It seems to be the choice for companies like Razer, Logitech, Microsoft, Kensington. There must be a reason for that. Can I ask why you don't like them? My question was if it was possible with some changes to the firmware. 

I'm new to this, recently ordered everything for Nrfmicro and looking forward to the build. Thanks for your work on this btw!

joric, 2019-12-16:
I use exclusively laptops since about 2006 and I don't have much USB ports but there's always bluetooth , so receivers are not my jam. I made https://github.com/joric/bluetosis specially to get rid of receivers and I'm not planning to bring them back.