joric/nrfmicro #18: The reset button to boot to the UF2 Mass Storage device.

luantty2, 2020-02-09:
Hi, I'm trying the nrfmicro but I noticed that there isn't a reset button on board, which might be needed in this process:
>After flashing the bootloader press reset button twice within 500ms interval to boot to the UF2 Mass Storage device. You have to convert standard .hex firmware to the UF2 format, using uf2conv.py.

Is that mean I need to create a button on my keyboard PCB? If it does,  which pins should  I connect to the button? Is that *GND* and *RESET* pin(P0.18)? 
Also, in file *custom_board.h*,  it says:
```
#define RESET_PIN GPIO(1,18)
```
I found that the reset pin of nrf52840 should be P0.18, does this matter?

joric, 2020-02-09:
Keyboards usually have reset buttons connected to the promicro reset pin. To reset the keyboard programmatically see https://github.com/joric/qmk/blob/nrf52-jian/keyboards/jian_ble/keymaps/default/keymap.c#L141 (it's a little bit complicated).

joric, 2020-02-09:
Regarding 1.18 yeah I guess it's a typo, should be 0.18. But it's not used anywhere, probably it can be remapped from reset to something else but I'm not really sure. It's connected to the promicro reset pin.

vexchow, 2020-05-07:
Hi Joric, 
Same question here.
I load the pca10056_bootloader-0.2.11_s140_6.1.1.hex or nrf51822_xxac.hex or s130_nrf51_2hex. 
And then connect reset and GND, But nothing happens.
[https://imgur.com/a/fniLNi7](url)

I spend 3 days to try, can you please give some suggestion?
Thanks a lot.

joric, 2020-05-07:
it's double reset. also check solder jumpers.

closed 2020-05-09 by joric