joric/nrfmicro #24: Parts list for v1.2

tlrogers-lap, 2020-05-14:
Joric, I just purchased 4 v1.2 alpha boards and am interested in building them. It does not look like the parts list BOM here [https://github.com/joric/nrfmicro/wiki/Hardware](https://github.com/joric/nrfmicro/wiki/Hardware) would align to the v1.2 alpha board. Might I get a parts list for the alpha?

joric, 2020-05-14:
You should've waited for the release ) There a new 1.2 pre-release that's different (no AP2112K at all).
Alphas should work though. You'd need a few parts, search for them on aliexpress or something:
* a midmount USB-C connector
* 10 uH inductor (0603 package)
* MCP73811 (probably will also work with an old MCP 73821, just use software pulldown for the PROG pin)
I guess that's about it.

closed 2020-05-14 by joric

tlrogers-lap, 2020-05-14:
Thanks a lot. Just found your github today and have been reading heavily on it as I am working on a keyboard and would like to leverage your amazing work.

Will the high voltage pre-release version still be running at 3.7 volts or was the high voltage reference going to a 5 volt version. Did not find enough info on the pre-release since you just posted it minutes ago ;^)

Trace

tlrogers-lap, 2020-05-14:
Will review the Kicad now and validate. May wait to build until you reach release. Thanks again for all the work!

joric, 2020-05-14:
High voltage refers to an internal nRF52840 LDO (1.8V internal current). 1.2 alpha uses internal LDO in conjunction with external LDO, 1.2 pre-release uses internal LDO and unfiltered battery/USB output, because, what the hell, all those pro micro-based keyboards are 5V and feeding 3.3V via TRRS is kind of questionnable (you could use 5V LDO there though).