joric/nrfmicro #27: I used the ZMK firmware ext_ VCC does not have the correct output

a826889260, 2020-09-18:
I used the ZMK firmware ext_ VCC does not have the correct output， When I'm not connected to RGB, the voltage is 3.3V. But as soon as I connect the RGB voltage to 2. + V。
is power pin problem？

joric, 2020-09-18:
yes if it's uninitialized it leaks through the floating gate

closed 2020-09-18 by joric

a826889260, 2020-09-18:
#if CONFIG_BOARD_NRFMICRO_13
    struct device *p0 = device_get_binding("GPIO_0");
    // enable EXT_VCC (use 0 for nRFMicro 1.3, use 1 for nRFMicro 1.1)
    gpio_pin_configure(p1, 9, GPIO_OUTPUT);
    gpio_pin_set(p1, 9, 0);
i try it 
but  it didn't seem to work

joric, 2020-09-18:
are you just randomly mashing keys
it's pin P1.09
so it should be GPIO_1

Lenbok, 2020-09-18:
Is nrfmicro directly supported in ZMK master, or do you need to build off joric's personal fork?

Lenbok, 2020-09-18:
Nice, I just looked and nrfmicro 1.1 onwards are in there now, (although I'm still on 0.03 boards until my 1.3 PCBs arrive.

a826889260, 2020-09-19:
There's more ahead   
 struct device *p1 = device_get_binding("GPIO_1");
But still no effect

joric, 2020-09-19:
maybe bad soldering. I got zmk working with stable voltage on the oled and leds. see wiki

a826889260, 2020-09-19:
I replaced the a03407 and found the job