joric/nrfmicro #36: BOM for v1.4?

pluskid, 2020-11-15:
Can we have a BOM for v1.4 in the same way as v1.3? I know that most of the parts are similar as v1.3, but I think copying over the old information and making necessary is very useful for beginners, esp. someone who has not been following along since the earlier versions. In particular, I was trying to follow using the v1.3 BOM and only half way through did I realize the component IDs are not one-to-one mapped. I have to cross check with the schematic to figure out which one is which for some of the components.

A few other minor suggestions: I think this figure from v1.3 is very helpful:

[image]

Can we have a similar one for v1.4 and future versions? It does not need to be based on 3D rendered images, I think annotation on real photo works equally well. 

The symbol on the silkscreen for the diode is not standard? -- or maybe I'm just not very familiar with the conventions. Maybe it helps to mention a word in each release page how to read the polarity? 

Thanks again for sharing such awesome project!

joric, 2020-11-16:
I don't have 1.4 pcbs and assembled boards it's in the status article, so maybe you could do it.

joric, 2020-11-16:
All done. Closing.

closed 2020-11-16 by joric

pluskid, 2020-11-16:
Thanks a lot for the updates! I'm happy to contribute, but I'm quite new to this and I'm not even sure if my first assembled board is working or not since I don't have something to burn the bootloader yet. It's great that you updated them!

pluskid, 2020-12-18:
The updated wiki is very helpful. Here are a few minor suggestions:

1. The BOM table list U3 and U4, I believe it should be U2 and U3?
2. The wiki says "You may improve the board using more expensive BAT60B and XC6220. See Components article.". As a beginner, I think the original instructions is more helpful. i.e. BAT60B schottky diode can replace 1N5819, XC6220B331MR-G can replace AP2112K, etc.

Thanks again!

joric, 2020-12-18:
fxd

joric, 2021-10-03:
New link https://github.com/joric/nrfmicro/wiki/BOM#nrfmicro-14