joric/nrfmicro #37: Able to flash bootloader using blackmagic and bluepill, but computer not detecting nrfMicro 1.1

harshitgoel96, 2021-01-30:
hi. i was able to flash the nrf micro using bluepill. there was warning, but no error. after flashing when i connect the micro to pc using micro usb, it is not getting detected by the pc. nor there is any led activity on the board.

harshitgoel96, 2021-01-30:
I am getting Vcc reading of around 3v over usb, so vcc, and GND are connected properly. could it be issue with data line under pads?

joric, 2021-01-30:
Check soldering or something. I don't have an universal solution for that. Check usb connector (it often gets shorted), and USB data (it's on the outer pads).

closed 2021-01-30 by joric

YHRen, 2021-01-30:
@harshitgoel96 can you direct me to how to flash the nrf? I have v1.1 soldered, but not sure how to flash it. Thank you.

joric, 2021-05-29:
@YHRen what the hell man. Did you miss the entire wiki https://github.com/joric/nrfmicro/wiki/Bootloader