joric/nrfmicro #38: Heating of nrfmicro

jrhrsmit, 2021-02-19:
In wake of your [reply](https://github.com/zmkfirmware/zmk/pull/669#issuecomment-780250351) on a PR I submitted on ZMK, I went on investigation.
It took me a while to understand what you were talking about (hence the confusing replies), but I think I understand your issue now, so I took the liberty of checking out your nrfmicro schematic.
I assume you were talking about D1 in the [schematic of the nrfmicro](https://github.com/joric/nrfmicro/releases/download/1.1/nrfmicro-1.1-schematic.pdf).
It seems D1 uses a SOD-123 package, and according to [this datasheet](https://datasheet.lcsc.com/szlcsc/1907060333_LGE-1N5819W_C402219.pdf) the 1N5819 has a forward voltage of ~0.4V at 0.6A (max LDO current, but 28 SK6812s may draw more than that), and with a typical thermal resistance of 250 degC/W, this results in a package temperature rise of 60 degC.
Thus it seems that it is indeed the cause of the heating problem you experienced.
The solution would be using a diode with a lower forward voltage and/or lower thermal resistance.

joric, 2021-02-19:
1) D1 is SOD-323 on all the boards except ancient 1.1 ones, not SOD-123.
2) Wasn't 1N5819 issue for me. BAT60 heats up too despite it's rated 3A. 

I've covered it here https://github.com/joric/nrfmicro/wiki/RGB#diode-overheating-issue but I'm really not sure which diode could solve the problem. Seems like pro micros don't have this issue and their power circuit is very similar.

jrhrsmit, 2021-02-20:
Ah forgive my shortsightedness. 
It indeed seems the BAT60 is the best JLC has to offer for SOD323 packages. However, I just checked out the layout, and it seems the diode has no copper pour/plane to dissipate power to (which can make a significant difference).
You could consider using a 4-layer PCB, using the middle layers as ground and VBUS (and connecting the VBUS pin with vias to that layer).
Another option could be increasing the copper area by using copper pours attached to the pads of the diode on the top layer. Other traces and vias should then be moved away as far as possible to increase the copper area. There are some whitepapers available on this subject.

Anyway, as I see that you've explored this problem already, I'll just close this issue.

closed 2021-02-20 by jrhrsmit