joric/nrfmicro #39: Question: nRFMicro 1.1 LEDs D2/D3 orientation

Raskilas, 2021-04-15:
Hi, my question might be silly and outdated, but which orientation should I use for LEDs **D2** and **D3** on **nRFMicro 1.1** board?

[image]

joric, 2021-04-16:
Are you kidding me. It's all there in the picture. Front has matching green on my picture and on yours. Bracket on the PCB is CATHODE (+), so put traces of the green shit there, don't look at the back, back is different from the front. Also See https://lighthouseleds.com/blog/polarity-guide-of-0402-0603-0805-1206-and-most-all-smd-leds.html

closed 2021-04-16 by joric

Raskilas, 2021-04-16:
> Are you kidding me. It's all there in the picture. Front has matching green on my picture and on yours. I only look at the front. Don't look at the back, back is different. Also see https://lighthouseleds.com/blog/polarity-guide-of-0402-0603-0805-1206-and-most-all-smd-leds.html Mind that LEDs don't have to blink by default, adafruit bootloader knows nothing about pin 1.10. The easiest way to check leds is probably Circuitpython, https://github.com/joric/nrfmicro/wiki/Circuitpython

Sorry for confusion, I meant where there is Cathode and Anode on board. It's NOT question about TOP/BOTTON orientation, it is about "Cathod on TOP of D2 or on BOTTOM and cathod on TOP of D3 or on BOTTOM". I just attached front/back pictures in case if someone in future won't be able to check back side of his/her LED.

joric, 2021-04-16:
There are standard footpints. Key (bracket) on the silk is Cathode (minus). It usually goes to ground via load resistor.

[image]