joric/nrfmicro #51: 1.8v in high voltage mode

ImMrMa, 2021-12-09:
Hi joric,
I follow your wiki made a board by myself, when use your bootloader, I found that in high voltage mode it can only output a signal of 1.8v. What should I do to output a 3.3v signal ?

joric, 2021-12-09:
https://github.com/joric/nrfmicro/wiki/Archive#power-switch-issues

closed 2021-12-09 by joric

ImMrMa, 2021-12-11:
Ok, So I'm using your nrf52 QMK branch, should I insert the code in QMK codebase somewhere is ok?

joric, 2021-12-11:
Unless you have a few years of spare time I don't recommend using wireless QMK, there are no maintainers. Try ZMK first.

ImMrMa, 2021-12-11:
Okay, thanks for your suggestion, although I have been trying on qmk for a long time, I will try zmk

ImMrMa, 2021-12-19:
Thanks again for the suggestion! ZMK has solved a lot of the problems I had on QMK. It was really cool!

ImMrMa, 2021-12-19:
By the way, I would like to refer mikoto as the first PCBA version, do you have any suggestion on the nrf52840 PCBA?

joric, 2021-12-19:
I don't give a fuck I know I was first.