joric/nrfmicro #55: single step debugging

billionfish, 2022-03-04:
Hi, I'm interested in your work, I made a keyboard based on your project.

Since I use qmk to compile the code, but I don't know how to debug it , can you give some advice on debugging?

For nrf52, I would like to know if there is a way to debug the code in a single step on IDE ?

Can this project be converted to a "SEGGER Embedded Studio for ARM" project for debugging? Or are there some tutorials that can guide me to convert the project for single step debugging in some IDEs like vscode?

I have successfully created a simple hello world using vscode's nrfconnect plugin and SEGGER Embedded Studio, but I don't know how to convert your project into these two IDEs.

Thanks!

joric, 2022-03-04:
I wish I knew. QMK is not under development in favor of ZMK, and there are network/lag issues. Try asking here https://github.com/sekigon-gonnoc

billionfish, 2022-03-09:
thank you for your reply

closed 2022-03-09 by billionfish