joric/nrfmicro #59: Black magic probe produced from ST-Link V2 requires different instructions

JiaGengChang, 2022-05-26:
Hi Joric,

I flashed black magic firmware into bluepill using ST-LINK V2 as the UART-to-USB method did not work for me (maybe because of the BOOT0 resistor issue you linked).

Now, black magic ports show up under "libusb-win32 devices" on windows 10 device manager, instead of showing up under Ports (COM & LPT).

[image]

I suspect it is because of this reason that the remaining instructions to unlock and flash do not work (error is that COM5 is not recognized).

The Ports (COM & LPT) dropdown only appears when I click "View -> Show hidden devices", and it is the following 
[image]

In this case, is there a way to locate this Black Magic GDB Server?

JiaGengChang, 2022-05-26:
Hi Joric,

As an update, I have also managed to flash black magic firmware into bluepill using UART and the official demonstrator GUI from STM. 

However, the issue that the black magic probe does not show up under Ports (COM and LPT) in Device Manager persists. It still shows up under "libusb-win32 devices", and so I do not know what its address.

JiaGengChang, 2022-05-26:
I managed to solve this problem. 

First, my version of zadig (2.0) was not the latest version (version 2.7). So I switched to that. 

Using zadig 2.7, I upgraded/downgraded both Black Magic GDB Server and Black Magic UART port to "USB Serial (CDC)". 

[image]

Two COM ports instantly appeared in windows 10 Device Manager. 

[image]

Selecting anything other than "USB Serial (CDC)", including the "libusb-win32" that is recommended in the wiki bootloader section, prevents the ports from being shown in device manager.

closed 2022-05-26 by JiaGengChang

Timception, 2022-10-25:
I managed to get to the same place you did. After that, I cannot get the bluepill blackmagic to flash the nrfmicro. Been trying for weeks. Im not a pro and the explanation and clip are too funny and skip so many things for noobs like me. I bought j-link, st-link, everything. None work after following the documentation. Help needed badly.

Sweet-tooth-man, 2023-02-15:
> Мне удалось добраться до того же места, что и вы. После этого я не могу заставить bluepill blackmagic прошить nrfmicro. Пытался неделями. Я не профессионал, и объяснение и клип слишком забавны и пропускают так много вещей для нубов, таких как я. Я купил j-link, st-link, все. Ни один из них не работает после выполнения документации. Помощь нужна очень.

did you manage to solve the problem?