joric/nrfmicro #61: Unrecognized Device Blue Pill to Blackmagic

drakumus, 2022-08-26:
Followed instructions to download blackmagic to blue pill:

- Hook up an UART adapter to Bluepill (A9 - RX pin, A10 - TX pin).
- Set Bluepill jumpers to the bootloader mode (BOOT0 jumper=1, BOOT1 jumper=0).
- Plug in UART adapter to the computer. Run "STM32 Flash loader demonstrator".
- Download to device blackmagic_dfu.bin at offset 0x8000000 with "Global Erase" option. < verified
- Download to device blackmagic.bin at offset 0x8002000 with "Erase necessary pages" option. < verified
- Put Bluepill jumpers back to the operational mode (BOOT0 jumper=0, BOOT1 jumper=0).
- Using Zadig after that to attempt to install libusb-win32 fails. < FAILS

When plugging back in the Bluepill windows no longer recognizes the device.
Attempting to install libusb-win32 drivers to the device with zadig fails as well.
This happened on 2 blue-pills.

joric, 2022-08-26:
Maybe this https://github.com/joric/nrfmicro/wiki/Bootloader#possible-issues

drakumus, 2022-08-26:
I went through those before posting:
The device can be connected overUART so that rules out the pull up resistor.
Windows doesn't recognize the device at all so the other issue referenced doesn't help me unfortunately. Their device just needed different USB drivers.

My running theory is the dfu flashed at address 0x800000 doesn't work for my board.

drakumus, 2022-08-29:
I bought another set of bluepills. This set matches the images on the troubleshooting page.
Checked the resistor value and it is the correct 10k.
Connected to the device over UART and realized something:
On both boards, iteration doesn't matter, it only detects 32kb of flash.
It looks like a minimum of 64kb is required for blackmagic probe.

drakumus, 2022-08-29:
[image]
[image]
A side by side of the 2 versions of blue pill, left one is the first set I bought, right one is the second set

I just noticed the component that has 32k written on it as well. Checking the aliexpress link you gave the first offering of blue pill on aliexpress also has that 32k:
[image]

drakumus, 2022-08-29:
Ok the issue is clear now.
The blue pills I bought off amazon (both types) use the cheaper STM32F103G6T6A (32kb flash)
The aliexpress blue pills use STM32F103C8T6 (64kb flash)

I guess let this be a warning that there are blue pills with smaller flash.

closed 2022-08-29 by drakumus