joric/nrfmicro #66: I'm beaten after frying 3 boards

klesh, 2023-08-03:
Hi, Joric, I'm amazed by your work, but building a development board is intimidating for me.
So I went and bought 4 boards from Taobao and somehow I managed to break 3 out of them 🥲....
What is done is done, but not knowing why is the thing that bugs me..😂.
I'm desperate for answers...and yet, google didn't tell me nothing, here is the only one guy that seems to be having the same problem https://devzone.nordicsemi.com/f/nordic-q-a/71117/nrf52832-gpio-irreversible-pin-damage-in-application-when-used-as-a-momentary-switch-sense-input

**I'm wondering if you could give me some pointers, **
I'm trying to build a Ferris/Sweep using ZMK direct pin mod, The left part is done and seemed to be working fine while the right part... not so lucky.💔
Here are the 3 broken ones so far...
[image]
I soldered the 2 on the right and found out each of them has 1 pin always grounded (actually 2, another one I never used also shorted ).  
So, I bought another one on the left and flashed the firmware, and used a tweezer to test the pins, it worked at first, and then again, it broke as well 😭...It seems like if the pin gets grounded too well for too long, it damage for good...

It would be great if you could take a look at my situation and suggest what I did wrong. 🙏
Thanks.

joric, 2023-08-04:
idk man. is it completely dead or just a few pins? when mcu burns it usually shorts vcc to ground

klesh, 2023-08-04:
@joric Thanks for the reply, it was a few pins.
I think it might be caused by testing the pin with a tweezer while the USB cable is connected.
I tried to solder a brand-new one without testing it, and it worked...

closed 2023-08-05 by klesh