joric/nrfmicro #68: How to check soldering for older versions?

ntrp, 2023-10-25:
Hello, thanks for this amazing project! I am having some issues with my board, probably due to bad soldering..

I have seen the https://github.com/joric/nrfmicro/wiki/Soldering#testing article but I have currently a older version (I think 0.03) which I built just now after ordering the pcbs some time ago. I tried the Diode testing mode between GND and any other pads but nothing happens, only activates when I touch the probes together.

Also I am using BMP over a BluePill and I can see this in the output after a `swdp_scan`:
```
(gdb) mon swdp_scan
Target voltage: unknown
Available Targets:
No. Att Driver
 1      Nordic nRF52 Access Port
```
Which looks wrong because I have seen other screenshots showing also a line saying `Nordic nRF52 M3/M4` and not only access port. Does that mean that my chip is badly soldered?

joric, 2023-10-25:
Tester should be set to diode mode not continuity mode. You measure voltage drop, not resistance.

closed 2023-10-25 by joric

ntrp, 2023-10-25:
Ok sorry.. stupid me, I switched the multimeter wheel to the right position but did not activate diode mode.. Also I was using the second pin, instead of the 3rd for GND, should doublecheck better next time..

I was trying to unlock using a BMP connected to SWC/SWD but when running the unlock code I was getting timeouts. 

What about the bottom pins? How should I verify they are in order? If all gpio pins have 0,7 vdrop they are ok?

p.s. I found 2 GPIO pins which are not connected so now I need to figure out which connection needs a touch up..