joric/nrfmicro #73: Really low battery life

madushan1000, 2023-12-15:
I'm using two nrfmicro controllers in [this](https://github.com/strayer/taira-keyboard) split keyboard. [This is my zmk config](https://github.com/madushan1000/taira-keyboard-zmk-config). The keyboards does not have any power draining things like RGB/displays, but the battery life is really low, about 3 days with 80mah 301230 lipo battery. 

Is this normal? any tips on how to improve it? Thank you :)

madushan1000, 2023-12-15:
I also noticed now that there is about 20% battery drain overnight if I forget to turn off the keyboard even though the deep seep timeout is set to 15 mins.

joric, 2023-12-15:
idk man. shouldn't be like this. try ZMK discord

closed 2023-12-15 by joric

madushan1000, 2023-12-16:
thanks, I will try that :)

madushan1000, 2023-12-16:
Just in case someone come across this. Turns out, my <keyboard>.conf file was in the wrong path and never got included(file has to be directly under config/). so probably sleep/idle modes never got enabled. Fixed it and waiting to see how the battery life is now.