joric/nrfmicro #74: Compatibility with IBM model M

IronLover64, 2024-02-29:
Hi, I want to use your circuit to control an IBM model M keyboard and I need to use all the available 28 GPIOs on the board as 24 will be used to control the keyboard and the other 4 for the indicator lights (num lock, caps lock, scroll lock). Will I run into any issues or make any compromises doing this? I remember reading that you can desolder the P0.00 and P0.01 crystal to use as a free pin, but would doing so essentially disable the bluetooth function? I am also wondering if it's not possible to use the P0.12 pin or any of the low frequency for one of the rows/columns . Thanks for reading.

joric, 2024-03-08:
Crystal is not needed, all pins are listed in pinout article. At least 28 pins (including xtal pins) should be available.

closed 2024-03-08 by joric

joric, 2024-03-10:
@IronLover64 https://github.com/joric/nrfmicro/wiki/Components#crystal