joric/nrfmicro #75: [Question] Pro Micro NRF52840 LED charging behaviour.

Finnitio, 2024-06-01:
Hey, could not find any other resources. Does anybody know the charging behaviour of the LED? It goes from solid to blinking, to off I think. But I do not know what that means? I also do not know why zmk gives 0 % charge over bluetooth but the keyboard still works for many days till the circuit shuts the battery off.

joric, 2024-06-02:
It doesn't go to blinking it's just on when charging and off when it's done. It blinks only if there's no battery. Measuring charging by voltage is often innacurate, last time I tried battery shuts off at 2.4V and ZMK may consider that below 0%. See https://github.com/joric/nrfmicro/wiki/Batteries about batteries.

closed 2024-06-02 by joric

Finnitio, 2024-10-14:
Anothe question, i noticed that when i try to charge my left side the whole keyboard does not work anymore. But for the right side it does noch affect the behaviour at all?

joric, 2024-10-14:
Separate halves should charge just fine, it should not affect the wireless link.