joric/nrfmicro #76: DFU in SuperMini

meekowai, 2024-11-13:
How do you enter DFU in SuperMini? shorting gnd to rst one or twice is not doing it. I read a drive in windows is supposed to pop up. Thanks.

gretel, 2024-11-13:
> How do you enter DFU in SuperMini? shorting gnd to rst once or twice is not doing it. I read a drive in windows is supposed to pop up. Thanks.

have a look at https://github.com/joric/nrfmicro/wiki/Bootloader

meekowai, 2024-11-13:
So i guess I have bad board. Blue led is blinking twice as fast as the red one. No constant light, even after shorting gnd to rst. Can connect to the board via BT from nRF Connect on iOS though.
I might need to try RaspberryPI route. Will it work or are the boards trash?

meekowai, 2024-11-14:
Was able to revive one board with RPi.
The only question I have is which bootloader should I really use?
Supermini nrf52840 says to be nice!nano clone, has the same pins on pcb. Logic says I should use nince!nano bootloader from adafruit (which worked in my case and allowed me to flash Meshtastic) but maybe it doesn't matter and I can use generic pca10056 or nfrmicro?
Sorry for dumb question but is bootloader, apart from everything else, responsible also for which pins are exposed? I understand Meshtastic has different variants of boards and now I wonder if good working order depends on bootloader/firmware combo or just pins assigned in the meshtastic firmware.
Thanks.

joric, 2024-11-14:
Defective board. Double reset should work.

closed 2024-11-14 by joric