joric/nrfmicro #77: pinout incorrect?

Finnitio, 2025-01-25:
[image]

I don't get this pinout would not work with zmk since some pins have no mapping for the nice nano v2. And the other Superminis have a different pinout.

joric, 2025-02-07:
It is manufacturers pinout. It also matches nicenano. 

[image]

[image]

closed 2025-02-07 by joric

Finnitio, 2025-02-07:
I opened the wiki to map the pins in zmk which did not work with the SuperMinis Pinout by pdcook. The Pins on the right side have a different order: 10, 16, 14, 15, 18 ... for nice nano and ProMicro as the alternative version.

The Supermini order is: 10, 11, 12, 13 ...

That confused me as I ordered a promicro as alternative.

joric, 2025-02-07:
I don't see either of those things anywhere on the pictures above. Quite the opposite, it is obvious that supermini and nicenano 100% match pdcook pinout. Maybe ZMK notation is different.

Finnitio, 2025-02-07:
It is still confusing if you do not need the alternative naming :)