joric/nrfmicro #80: ProMicro battery readings

richonguzman, 2025-04-28:
Hi,

don't know where to ask this but your repository helped me a lot to start.

I have ProMicro NRF52840 successfully advertising and using 700uA when coding on Platformio/VSCODE.

should i expect lower values with this board disabling parts of the board and such?

joric, 2025-04-28:
700uA spikes is OK for radio I guess, even 1-1.5 mA is fine. As for AIN02 I don't think it's connected to anything on nicenano clones. You just read internal VDDH voltage, refer to ZMK. I don't think there's a way to read battery voltage when USB is plugged in.

closed 2025-04-28 by joric

richonguzman, 2025-04-28:
thanks for the info,

so the multi tester is showing the 700uA spikes when BLE advertising! interesting!

AIN02 (pin4) does not show me any battery voltage readings running it without usb connected. Besides classic pin initiation should i declare any specifics to make it read the battery voltage?

joric, 2025-04-29:
I just said it's not connected to anything.